moonglow/FlashForge_Marlin #137: ver 2.0.9.5 wont turn on extruder fan #2

fbb034, 2023-01-19:
I recently upgraded to 2.0.9.5 and I noticed that the 2nd extruder fan stays off. I have checked my HW and the fan is good but it does not get 24V to turn on. The 1st fan works perfectly. I'm using "dreamer_2.0.9.5_dir_swap_0115202"

fbb034, 2023-01-25:
**UPDATE**

I have looked into this in much more detail and I discovered that the problem is NOT due to your new released SW.

I first downgraded to the previous version and the 2nd extruder fan still didn't work, so I disassembled the printer and found that Q601 was damaged. Q601 is a FET that is used as a switch for the fan. The damage was visible with the naked eye.

[image]

Since the part markings were no longer visible I read it from the identical part that is used for the 1st extruder fan. This part is Q603
[image]

The part marking (WKC-09) turns out to be a N-channel Trench MOSFET (https://www.mouser.com/datasheet/2/916/PMV20EN-2938893.pdf)

After replacing this part both fans now work perfectly.

So why did this happen...

It turns out that I had mistakenly installed a 12V fan instead of a 24V fan. The 12V fan worked for a long time and just coincidentally damaged the FET after I upgraded to your latest version.

Hopefully these details will help someone having the same issue.

closed 2023-01-25 by fbb034

moonglow, 2023-01-25:
@fbb034 thank you so much for your investigating and tips !!!