moonglow/FlashForge_Marlin #138: 2 color printing issues

digital0785, 2023-01-20:
not sure if i'm doing something supremely dumb or what.. but when i try to do 2 color printing a few things happen. 1 it puts the first layer down with the  "correct" extruder, however after that it doesn't change extruders like it should it prints the whole thing with the other nozzle mind you that should only be used for the inner part. 

additionally it keeps stopping right over the part when it "should" be switching nozzles and purges a bunch of filament which I'm assuming is from having the .25 mm nozzle switch prime amount .. but like i said it does actually switch nozzles so it stops at the corner of the model purges a big blob .. then moves to the prime tower does its loop and goes back to printing with the same nozzle 
[image]
[image]
[image]
[image]

[image]
[image]
[image]
I have each set to the appropriate nozzle to use but it doesn't seem to care.. the gcode preview in cura shows it being the appropriate nozzles (i have the one nozzle printing at 60 vs the other printing at 80 so it's a very obvious difference) but best it does is print the prime line.. the appropriate part of the brim and first layer in the right extruder but it never goes back the only t0 in the gcode is the initial extruder selection then the cool down at the end

sorry for the bother trying to get it dialed in and this was the last thing I needed to calibrate and i've it a road block.

digital0785, 2023-01-21:
it's definitely something with cura superslicer works fine .. hmm I'll ask that group again

closed 2023-01-23 by digital0785

Spinboost, 2023-02-05:
What did you do to solve the problem of it switching extruders and putting a blob down before going to the prime tower?