moonglow/FlashForge_Marlin #142: FF Firmware Writer Failing

RexVaga, 2023-02-15:
I believe I'm following all of the steps properly, but when trying to flash one of the Marlin firmware versions, the FF Firmware Writer will have the following issue: 
[image]

RexVaga, 2023-02-15:
When using the 08_10_2021 version of FF Firmware Writer, I have this issue:
[image]

capitaotmn, 2023-03-14:
Have you tried to load the file, click on start and restart the printer?
Worked for me

zenmediasocial, 2025-01-19:
I have the same problem with my Dreamer NX as you do in the first screenshot. Were you ever able to resolve this issue? 
I get a beep then a really long beep and the NX restarts.

DUFF1856, 2025-04-01:
Same issue on windows 11. Tried starting flash at various times during boot but always fails and beeps short and or short and long. 

Anyone find a solution?