moonglow/FlashForge_Marlin #146: Flashing New STM32 for MightyBoard

Printo1987, 2023-03-10:
Hello I see the Marlin page for Dreamer is growing and a new marlin has been added.

Last week I burn my cpu from mightyboard and replace for new. After I use Original Bootloader from old CPU to new CPU by STLink.
I turn on printer everything ok but I only see screen Blue with Logo and on the left corner i see writing "check ID fail" 
I use too FF Flashing Tool end encrypt code my original Bootloader firmware, but not working after this step Screen is Black.
Anyone can help me how fix it ?

moonglow, 2023-03-10:
Hi @Printo1987 !

Original FF bootloader check internal hash inside firmware ( based on MCU UID ). So that mean what it will NOT WORK on another MCU ( because each of them has their own UID ) , but i and @tracedgod ( i think his version is better ) published patched bootloader, what you can use to flash to your new MCU.

https://github.com/moonglow/FlashForge_Marlin/issues/121#issuecomment-1250384307

Printo1987, 2023-03-11:
Workink thank you for solution :) Great Job Guys;)

P.s  You plan add ready pre set from Marlin 2.1 ?  becuase I see few days ago you update Marlin 2.1 but full pre set tags is only for Marlin 2.0.X  [v0.15.1] (https://github.com/moonglow/FlashForge_Marlin/releases/tag/v0.15.1).  Will be fine if you make this same for Marlin 2.1.X :)

moonglow, 2023-03-12:
Marlin 2.1.x for now buildable not for all configuration, i think we will tune it and release soon.

closed 2023-03-12 by moonglow

jerstyg, 2023-07-19:
> Marlin 2.1.x for now buildable not for all configuration, i think we will tune it and release soon.
hello Can you help me iwas trying to flash my dreamer nx a marlin firmware and accidentaly flashed a wrong firmware Dreamer instead of NX :( my printer now stuck on logo flashforge how to fix it? thanks