moonglow/FlashForge_Marlin #159: 🐛 [BUG] - <ff_fw_writer will not launch>

vr3zz, 2023-09-28:
### Description

it will initially throw an error saying that it cannot find the msvcr120.dll file: 
[image]

then after copying the dll to the folder it will instead throw this error message instead: 
[image]

After doing windows updates, running sfc /scannow, and redownloading the file multiple times I still cant seem to get the flasher to launch. 

### Reproduction steps

```bash
1. Download from GitHub
2. extract files 
3. run the application with the dll files copied into the folder
```

### Printer

FlashForge Dreamer

### Marlin Version

N/A

qwertychouskie, 2023-09-28:
Remove the manually copied DLL, then install https://www.microsoft.com/en-us/download/details.aspx?id=40784 (both the x86 and x64 versions).

vr3zz, 2023-09-28:
Ah that was it, I had the x64 version installed but not the x86

closed 2023-09-28 by vr3zz