moonglow/FlashForge_Marlin #163: Bought broken FlashForge Dreamer any ideas ?💡 [REQUEST] - <title>

xromansx, 2024-03-15:
### Summary

Bought this printer with long motherboard power turns on I see some of the leds turs on on MB, not detected when usb pluged boot0 with vcc doesn't put it in DFU. Next thing want to try st-link but need to find it before. 
I see that controller has 3v on vcc so power goes to it, wanted to ask what has 16GB microSD on it because mine is empty. if stlink doesn't help I'm planning to resolder STM chip. 
Does someone has any experience recovering this type of machines ?
Thanks 

### Basic Example

Recovery broken MB

Mrnt, 2024-04-28:
Micro SD card slot is for internal (print file) memory and can be used to update the firmware. It should not matter if the slot is empty.
Could be firmware is bricked in which case you will prob need stlink to refresh the firmware.

xromansx, 2024-05-02:
I resoldered the CPU but can't find anywhere backup of original STM dump, if someone can please share it would be really nice. Thank you.

xromansx, 2024-05-03:
Used [tracedgod](https://github.com/tracedgod) bootloader everything works. Thanks.

closed 2024-05-03 by xromansx