moonglow/FlashForge_Marlin #165: Replaced CPU looking for full dump to recover.

xromansx, 2024-05-03:
### Summary

Flashed fixed_dump that I foung here through stlink, was hoping that it boots but no luck. Tried to FF Firmware Writer after got following messages:
Start control
CMD M601 Received.
Control Success.
ok
CMD M601 Received.
Control Success.
ok
Create file failed
Stop

**When select trigger only receive:**

CMD M601 Received.
Control Success.
ok
CMD M601 Received.
Control Success.
ok
succeed
ok
succeed
ok
Stop

run out of ideas, would be glad if someone has full dump or can make one. 

Thanks.

### Basic Example

Full dump.

xromansx, 2024-05-03:
saved fw in sys folder on SD could flash but printer still doesn't boot.
[image]

Maybe there is some sort of CPU validation in FW.

xromansx, 2024-05-03:
Used [tracedgod](https://github.com/tracedgod) bootloader everything working perfectly, Thank you.

closed 2024-05-03 by xromansx

dnnp2011, 2024-10-13:
For Posterity, I was having this same issue with a Monoprice Inventor 1 (rebranded FlashForge Inventor 1).
Tried flashing with OctoPrint Firmware Updater, STM32CubeProgrammer over USB & UART, FlashPrint, FlashPrint-MP, and ff_fw_writer tool. Had some issue or another with all of them. ff_fw_writer would seem to work but just restart into old firmware (same problem as @xromansx)

In the end, I flashed the firmware from my Raspberry Pi running OctoPrint (externally powered) using @tracedgod fork, and it worked successfully!

valik1017, 2025-03-25:
> Used [tracedgod](https://github.com/tracedgod) bootloader everything working perfectly, Thank you.

Hello!

Thank you!

Can you tag please this bootloader?

Thanks

valik1017, 2025-03-26:
> Was quite a long time ago. Check maybe you can find in his repository.

I searched, but I didn't find anything.
Anyway, thank you.