moonglow/FlashForge_Marlin #166: 🐛 [BUG] - <title> Dremel 3D20 cannot flash firmware

Fred583, 2024-05-26:
### Description

I have been trying to fix this printer with no luck. I cannot flash as the previous flash got corrupted and now I cannot get past the build on screen. I have tried to use an ST-Link, but I cannot find the bootloader byte that needs to be changed from 0x00 to 0xFF in order to update from SD. I can flash firmware through the ST-Link, but after restart I get the same problem.

### Reproduction steps

```bash
I have no idea how to explain reproduction
```

### Printer

Dremel 3D20

### Marlin Version

2.1.2.1

Fred583, 2024-05-26:
I can upload whatever file is needed, or if I need to flash a new bootloader that works too. But right now I'm running in circles trying to fix this.

Fred583, 2024-05-28:
Okay just to continue the troubleshooting process, came across another issue that had the unlocked dreamer bootloader in it. Flashed it and tried every update process I could, and only got a white screen. So as of right now I'm still stuck with the backup of my locked bootloader with no idea where the trigger byte is.

OneMonkeyArmy, 2025-02-10:
Sorry to answer with many months late... but if you want your printer back to life... I know every single bit of those boards....

I have revived dead (BRICKED) flashforge boards with a success ratio of 99.9%... 

Even have a modified version of Bootloader for this particular problem...

Contact me on my emails:

daniel.massicotte@yandex.com
druidfirmware@gmail.com
or
dvamontreal@hotmail.com

Fred583, 2025-02-20:
> [image]
> 
> The bytes you are looking for are at the end of those 2 lines (address in HEX on the left)
> 
> But I am pretty sure that you did the mistake of erasing your unique encoded key that was written "just after flashforge12" and just before the 4 last bytes...
> 
> So you will need to use a modified bootloader, that can work with any firmware, but it no longer perform this test, that prevent the firmware to load at boot... 
> 
> Contact me thru one of my emails (previously listed)... if needed I will send you a simple installer that will correct everything in your dead printer...
> 
> 

I just emailed you. I appreciate your response!

OneMonkeyArmy, 2025-03-02:
I have search my emails, but never saw yours... Anyway, what have you done and what do you need...  I can solve any problem on a 3D20 ...